don/BluetoothSerial #431: Method discoverUnpaired is not working in background apps (android 10)

nasasutisna, 2020-09-29:
When apps is running background apps, the method discoverUnpaired is not working in android 10. I tried it on Android 8.1, running normally.

Please help, thank you

WaldiirNeto, 2020-10-10:
I have the same problem friend !!

WaldiirNeto, 2020-10-10:
[image]

only this method that is causing a problem, the capacitor simply kills the bluetooth event and waits for something that has already been killed

aavelyn, 2020-12-03:
Hi, we have the same issue with Galaxy Tab A7 and Android 10. Location Permission is granted.
But neither does `discoverUnpaired` nor `setDeviceDiscoveredListener` emit any values, while another tablet with android 8 or 9 work fine.

Does anyone has a fix or workaround?

fappels, 2020-12-14:
Most Android 10+ devices also need Manifest.permission.ACCESS_FINE_LOCATION