don/BluetoothSerial #432: Not working at all

LeoMH, 2020-10-08:
I try to connect to an ESP32 using your plugin. But every time I use a function of the plugin, the code seems to stop working and neither of the callbacks gets called. One example would be this: bluetoothSerial.connect("24:6F:28:7B:D8:76",success, failure). The app stops there and no callback gets called. I can't figure out what my mistake is.

philcom1, 2020-10-08:
Are you using an Android or iOS device for scanning and connecting?

LeoMH, 2020-10-08:
Android

WaldiirNeto, 2020-10-10:
are you using android 10? this plugin is having problems in this version of android

LeoMH, 2020-10-10:
Thanks for responding. I got it working. I had to remove android and add it again.

philcom1, 2020-10-10:
Nice! Which @ionic-native/bluetooth-serial version are you using? Maybe it could also be this problem ;)

LeoMH, 2020-10-10:
It probably did not get added to the platform when I installed it. I used the bash command to add it, so I guess it was the most recent release.

WaldiirNeto, 2020-10-12:
whoever has this problem, it could probably be that you have disabled the option to not allow access to the location just in the use of the app, let it always use the location.

WaldiirNeto, 2020-10-12:
I don't think there's a problem with a plugin, but with the mobile settings

aavelyn, 2020-12-03:
Hi, we have the same issue with Galaxy Tab A7 and Android 10. Location Permission is granted.
But neither does `discoverUnpaired` nor `setDeviceDiscoveredListener` emit any values, while another tablet with android 8 or 9 work fine.

Does anyone has a fix or workaround?

inlacou, 2021-01-11:
Created a pull request that should solve this issue.

> are you using android 10? this plugin is having problems in this version of android

The issue on Android 10 and above is that ACCESS_FINE_LOCATION is needed instead of ACCESS_COARSE_LOCATION.

Lucas-Wanat, 2022-08-30:
> quem tiver esse problema, provavelmente pode ser que você tenha desabilitado a opção de não permitir o acesso ao local apenas no uso do app, deixe ele sempre usar o local.
How could I do this? I still don't quite understand.