don/BluetoothSerial #438: I cannot print the PDF file correctly in my project

adilson181, 2021-03-01:
Hello, I have a problem printing the PDF file in my project, it is printing with strange characters, could someone help me?

This is the result of how it is being printed: 

[image]

I'm using this plugin to convert from base64 to ArrayBuffer [base64-arraybuffer](https://www.npmjs.com/package/base64-arraybuffer
)
Below is the code I am using:

print.service.ts:

```
import { decode } from 'base64-arraybuffer';

@Injectable({
  providedIn: 'root'
})
export class PrintService {

  constructor() { }

  base64ToArrayBuffer(base64) {
  //This convert string base64 to ArrayBuffer
    var converted= decode(base64);
    return converted;
  }
```

config.page.ts:

```
import { Component, OnInit } from '@angular/core';
import { PrintService } from 'src/app/services/print.service';

@Component({
  selector: 'app-config',
  templateUrl: './config.page.html',
  styleUrls: ['./config.page.scss'],
})
export class ConfigPage implements OnInit {

  bytesConverted:any;

constructor(private print:PrintService) { }

convertBase64(){
    const base64 = "data:application/pdf;base64,JVBERi0xLjMKMyAwIG9iago8PC9UeXBlIC9QYWdlCi9QYXJlbnQgMSAwIFIKL1Jlc291cmNlcyAyIDAgUgovQ29udGVudHMgNCAwIFI+PgplbmRvYmoKNCAwIG9iago8PC9GaWx0ZXIgL0ZsYXRlRGVjb2RlIC9MZW5ndGggNDA2Nj4+CnN0cmVhbQp4nI2bzXLbVhKF93mKu5hFUhVD+CUAVc2CkShHiSU6Eu3FVDa0RNucSKKHouOpvG4WqaznBaZxu2/3aTiQk1Qs8ZyDS/iyG/0Fgsrww1d51rThE33J8zw8/6qIX/HP/Tsyy44y363C0VkROpLD6m1YrL76T+Dj3gXyqi6r6lCXZVZWYXUbvp5fzsNZFv518v3VcvXjIlzMr1bnl+F6efV6sVpcnc+/Cat/0zLhJ79Q3WQ1LVTUaaGTy5c/HIeyotNosrzrjyhaPOuacL44Dn3ettWsaMt8YrlZlc3aUOczPa/X4WTxIixX89fny7BaXq/m34aibSeOL/Imq6pQ9Z2ez+bhsN9NxPt6OPuqz1P6+av5+dX82csrOyDP6qqg3ZaNbYYTrLo669pwT/vb5OnVXbj+Cl/fh6IYdhnsosxoJ8ynxXrno0A+LVeiX9K5V+bTx9e1zkeB/CZr3emVbVYW5le0XIO+E8in5Trn11kF51e1w76hj8I9fY5ZPkO/LoYKNp+2v3A+CuTTcm7/6i5rZ+Y3tJzbPyeQT8u582uarK/Bp+Xc/jnhPsyKbObOj8qzKMGn5dz+OYF8Ws6dX5tnFdRHS8u583MC+bScO7+2zerO/I6Wc+fnBPJpOXd+9N0M9q9rx+Xbuvqlvqjc+fVF1sH+UfvMXH04gXxazp1f32c57F+R03quQLwyJOpxiwzXBuyxnNb0TeaUoQuLUZsUReXquChozdInUBkS/aiUizJ3tVzQJa/OfQKVITEblXNB/dji9YD6r3Eb6pUhQWv6M6We7CtMtNAjnGhd1xTUljN/pnWZFQUmaugTTtSucwpqzdafad1nJe5pU46K2ytDgtb0Z9rMshr3tOlHBe4VSsxoTX+mwwjBPZ3Rmr6CnDIkaE2/p9SnHZ5pW8FVnxOVmwNFOxu1UtF2vtapNzvXjF4ZEtW41mmEulrvOrj+c6JzE6Gglh3Vel9mFVZQ38AM4ETjpkJBw9OPhYL6toEKKvMS+ocHU+k6qqQ57EdDSX3b1pjooX840buOKmla+vFQUt/2JSZmo1r3ypDoRyMiwZMAUjsBSDQ6O9p5KgS6lAxkcLq7+XhPLLEL84//3d5t1/twuw6Xu8M6nG0fb9Z34XYTTnYPjx/vt7e7fVjcbYg7HrY36ylyquOsoHKVd7j8fRc+bPb328MmrD/sd79utoc1vyUtfbP/83bL356fXFx/gU+afLiMJj7hV8YnyRU+MVv4RH3GEfBRUD4xX/gk+YIj4KOgfAI+80nyBUfMd4LyCfjMJ+ozjoCPgvKJ+cIn6jOOgI+C8gn4zCfJFxwx3wnKJ+Azn6jPOAI+Cson5gufqM84Aj4KyifmC58kX3AEfBSUT8BnPtHqYxyB8kNB+QR85hP123H5tq5+hU/MFz5Rn3EEfBSUT8BnPtH6FhqBBnCK8QkmmE8swTSCCVSMTyAhfGIJphFMoGJ8gq2cu1pONIIJVIxPMMF8ogmhEUg4xfgEE8wnlmihR5BPNCF8AgnhE0vU0CfIJ5ZgPsEE84ld28pRcXvF+AQTzCeW6EcF7hXjE0gIn1iCaQQTqBifQEL4RBNCI5io3BxIfIKJzte60AgknGJ8gonG17rQCCY6NxESn0BC+MQSDcwA5BNLdKOxkPhE54rQCA6m0nVU4hNMMJ9Yoof+QT7RhPAJJIRPLDEb1bpXjE8skfjEsQJRCoXqagjFmxr/u92+m7qpQVNxaAVLn24eb/bbP36fOqAo+GO0I3463G6mwhWFZxh+dTkVbfgiZdHXd+HVw/YwladSIIbx+RUR1t3EAbwt1SzFp+4KyY5Y8HK5moez8+uT+YuwuDhfnZ9O3d+SQ8s+Hbq4CPOL784Xl6tFOF2E75cXk7s6i8PW3rX4Np+675V21cI/Pp+KDpcHt26ZTy9c5sw4fy/NG1rQtbOM2frpXbHgxfbul3D9fv3LZNnIbtghf2M3LDxdY7IbsG755d2YTP8lTFN70CxJMM2vDKaTKzBttsC0+szO4KOgMG2+wHTyhZ3BR0FhGnyG6eQLO5vvBIVp8Bmm1Wd2Bh8FhWnzBabVZ3YGHwWFafAZppMv7Gy+ExSmwWeYVp/ZGXwUFKbNF5hWn9kZfBQUps0XmE6+sDP4KChMg88wrdXH7Azlh4LCNPgM0+q34/JtXf0KTJsvMK0+szP4KChMg88wrfUt6AwN4BSDaUwwTFuC0RkTqBhMQ0Jg2hI8lTCBisE0tnLuajmhMyZQMZjGBMO0JgSdIeEUg2lMMExbooUeQZjWhMA0JASmLVFDnyBMW4JhGhMM03ZtK0fF7RWDaUwwTFuiHxW4VwymISEwbQlGZ0ygYjANCYFpTQg6Y6JycyDBNCY6X+uCzpBwisE0Jhpf64LOmOjcREgwDQmBaUs0MAMQpi3RjcZCgmmdK4LOOJhK11EJpjHBMG2JHvoHYVoTAtOQEJi2xGxU614xmLbE6Gbf1E9DGXvyWOUJf8NhAM/hdtv2sHl4nAIKGeJ2bPkUWpW0t5J7vb7b7Zluw9U/JnmlGJaHw6on6EbeokjZgft3w+3DJ9ZveH09xsNh3LeSLrtPbVzZFf4vtQ4v1+/W++m3LWhWFO7A6ssYVlKP0bVPMExeKYapyxgGNmOY+fw3Bh+FhGHgM4apz9SFPgoJw9CPGKY+Uxf4TkgYhn7EMPMjdaGPQsIw8BnDzI/UhT4KCcPQjximPlMX+E5IGIZ+xDDzI3Whj0LCMPAZw8yP1IU+CgnDwGcMU5+pC30UEoahHzHMqi9SF5YfCgnD0I8YZn47Lt/W1S9jGPiMYeZH6kIfhYRh6EcMs/pm6MIGcIpimEtEDINEhC6XQEUxDBOMYZCI0OUSqCiGuVbOXS0LdLkEKophLhExzBIMXZhwimKYS0QMg0QLPQIYZgnGMEwwhkGihj4BDINEHDIuETEMrm3lqLi9ohjmEhHDINGPCtwrimGYYAyDRIQul0BFMQwTjGGWYOhyicrNAcEwl+h8rTN0YcIpimEu0fhaZ+hyic5NBMEwTDCGQaKBGQAYBoluNBYEw2yuMHS5wVS6jhIMc4mIYZDooX8AwyzBGIYJxjBIzEa17hXFMEj8zZ+5Mk20xXDYAAVny6uLeXg5fz6/WFyulpN3gGr+SPW41/MXy6vhuOU0gvB7zep0zOn24f1mO/mgWJlXsRftgC+DWNOm7Gq/u3mSwtrhMTQ4IP8yDTXxXmmiIX5lNJRcoSGzhYbUZ/gBHwWlIfOFhpIv8AM+CkpD4DMNJV/gx3wnKA2BzzSkPsMP+CgoDZkvNKQ+ww/4KCgNgc80lHyBH/OdoDQEPtOQ+gw/4KOgNGS+0JD6DD/go6A0ZL7QUPIFfsBHQWkIfKYhrT6GHyg/FJSGwGcaUr8dl2/r6ldoyHyhIfUZfsBHQWkIfKYhrW9hH2gApxgNYYJpyBLMPphAxWgIEkJDlmD2wQQqRkPYyrmr5cQ+mEDFaAgTTEOaEPaBhFOMhjDBNGSJFnoEaUgTQkOQEBqyRA19gjRkCaYhTDAN2bWtHBW3V4yGMME0ZIl+VOBeMRqChNCQJZh9MIGK0RAkhIY0IeyDicrNgURDmOh8rQv7QMIpRkOYaHytC/tgonMTIdEQJISGLNHADEAaskQ3GguJhnSuCPvgYCpdRyUawgTTkCV66B+kIU0IDUFCaMgSs1Gte8VoyBKOhsrJm1LN8Ah9KKnsS75HEh8uuztswofN3TqcvF//uhluT81vNo+Pu7C5Px7dt5kGLTqXgQqGWclLvz8cPhwfHX369Cl7u/5t83C7zj7ss3e7X7M3+6OHtzeboxt+86mH2egyMqMV6Uvd8E8OC/pA83L4BDoqy66nZE7VRBNr+G2F4UD6rwpNU/Q05YeZVhWzLzBPGSEsMQ+/MuZJrjCP2cI86jPigI+CMo/5wjzJF8QBHwVlHvCZeZIviGO+E5R5wGfmUZ8RB3wUlHnMF+ZRnxEHfBSUecBn5km+II75TlDmAZ+ZR31GHPBRUOYxX5hHfUYc8FFQ5jFfmCf5gjjgo6DMAz4zj1YfIw6UHwrKPOAz86jfjsu3dfUrzGO+MI/6jDjgo6DMAz4zj9a3EA40gFOMeTDBzGMJJhxMoGLMAwlhHksw4WACFWMebOXc1XIiHEygYsyDCWYeTQjhQMIpxjyYYOaxRAs9gsyjCWEeSAjzWKKGPkHmsQQzDyaYeezaVo6K2yvGPJhg5rFEPypwrxjzQEKYxxJMOJhAxZgHEsI8mhDCwUTl5kBiHkx0vtaFcCDhFGMeTDS+1oVwMNG5iZCYBxLCPJZoYAYg81iiG42FxDw6V4RwcDCVrqMS82CCmccSPfQPMo8mhHkgIcxjidmo1r1izGOJv3yqraxjpxZ1VuRMOMvL61cX56fLq/AsnLw8C3lPH3A9rNI/o+b5Z7g8e7ZIT2zR1ykaaeLAyNu08OiJreWL5fP5yXxJ73K9uAh8v4gfB5t8fCmPPZLnaclvw7PPfk42hXJlN0y0oSbk4Muzk014yAYEGiioacL1n/vthl4PtHmUl0f00dIRxXGeHzftZ2A3+Wud7VAPRac7+nK/O+xudnfx9wbmHw+7/fa39R+/744DXUzovYp4Ai0xWDOb+KsP29/EVpM1T9eH9efLfX7a/dM0V8wiZwjNySulOXWZ5sBmmjM/whv6KCSaA59pTn2GN/RRSDSHfqQ59RnewHdCojn0I82ZH+ENfRQSzYHPNGd+hDf0UUg0h36kOfUZ3sB3QqI59CPNmR/hDX0UEs2BzzRnfoQ39FFINAc+05z6DG/oo5BoDv1Ic1Z9Ed6w/FBINId+pDnz23H5tq5+mebAZ5ozP8Ib+igkmkM/0pzVN7MbNoBTlOZcItIcJCK7uQQqSnOYYJqDRGQ3l0BFac61cu5qWdjNJVBRmnOJSHOWYHbDhFOU5lwi0hwkWugRoDlLMM1hgmkOEjX0CdAcJCLNuUSkObi2laPi9orSnEtEmoNEPypwryjNYYJpDhKR3VwCFaU5TDDNWYLZzSUqNweE5lyi87XO7IYJpyjNuUTja53ZzSU6NxGE5jDBNAeJBmYA0BwkutFYEJqzucLs5gZT6TpKaM4lIs1Boof+AZqzBNMcJpjmIDEb1bpXlOYgYTRHEJG1VcjpX/mW2iDePRkuYjf34ei8CKe7L9+S4jtHdG50seMfmW3fEF88xsewto/h/OFme7t5OGwew89fv9hsw9nmdrNf39EVgbakIPAoyp+/OQ5X/wjPhn9GyFSF2cQb0/WVCqYZICMi6N3u4+3p2SZcbx8Pm/v1IzEi3B27Gexnt2832c3uPnuz//xRqdq/T90W7s99Yqr46HE5fBTD2z6NoROLMJJSpckii/vtYXu7Dpv7sL5/sx12K4x/j+Ovl2rr+H+iNNpkqfe7e4LGd+ubNfxE9P+mDSShCmVuZHN0cmVhbQplbmRvYmoKMSAwIG9iago8PC9UeXBlIC9QYWdlcwovS2lkcyBbMyAwIFIgXQovQ291bnQgMQovTWVkaWFCb3ggWzAgMCAyMjYuNzcgNDM1LjkwXQo+PgplbmRvYmoKNSAwIG9iago8PC9UeXBlIC9Gb250Ci9CYXNlRm9udCAvVGltZXMtUm9tYW4KL1N1YnR5cGUgL1R5cGUxCi9FbmNvZGluZyAvV2luQW5zaUVuY29kaW5nCj4+CmVuZG9iago2IDAgb2JqCjw8L1R5cGUgL0ZvbnQKL0Jhc2VGb250IC9UaW1lcy1Cb2xkCi9TdWJ0eXBlIC9UeXBlMQovRW5jb2RpbmcgL1dpbkFuc2lFbmNvZGluZwo+PgplbmRvYmoKNyAwIG9iago8PC9UeXBlIC9Gb250Ci9CYXNlRm9udCAvVGltZXMtSXRhbGljCi9TdWJ0eXBlIC9UeXBlMQovRW5jb2RpbmcgL1dpbkFuc2lFbmNvZGluZwo+PgplbmRvYmoKOCAwIG9iago8PC9UeXBlIC9YT2JqZWN0Ci9TdWJ0eXBlIC9JbWFnZQovV2lkdGggMTgwCi9IZWlnaHQgMTgwCi9Db2xvclNwYWNlIFsvSW5kZXhlZCAvRGV2aWNlUkdCIDEgOSAwIFJdCi9CaXRzUGVyQ29tcG9uZW50IDEKL0ZpbHRlciAvRmxhdGVEZWNvZGUKL0RlY29kZVBhcm1zIDw8L1ByZWRpY3RvciAxNSAvQ29sb3JzIDEgL0JpdHNQZXJDb21wb25lbnQgMSAvQ29sdW1ucyAxODA+PgovTGVuZ3RoIDQzND4+CnN0cmVhbQpYhdWXy7HcMAwEkQHyz3IygNFD8n2qfPP4sFpJKzUPKPyGUNWnHzOjmq69SSXJIMfFuc9cu8piRfk+qYerpjDupTAXpGrX/hNfp/Q3u//O98QvYta2+x3PBCfNv44f9ZDgt0ZbvZb7kRxfO+vTpgTHtrx0zca4aq2tVXKD5fUsyZ2aTchsvn2vet5F+MJtawA+ua6iHM/wi7Pp7K/kZPg+YrdfUdWtqhRvQuX3dneP3Boxjp0jGbrqp0pyysqqsQkn8ZbwILcXTVtgtQhgKcjHKaH3htYjiLfXM5xa4jc4OG70VpDztmvDUiGr/Ac5xtzUtkfujzClON2Ac8Rtl+qpdorTeGw4zgjSVFUK8na8nmS3GzDJC92QIzfepcPc2zA5sfT13fVznC1fxO1MXDx0kuOG5cKZh/XVvQxXu5iazc2SQWqS3Krnrdlt5yEpyS3afQSD2N2wxfjGCatujvHGoyR/ve35Ec9ePYf4+e44tXuG4KfnGS6Pdn3qSSdBSvKjqH0GjDM25jlyZ4GywMa5faPH546OQV5Hjk673RkvyMeGmF0sUf42S/LPPv4ABpibXgplbmRzdHJlYW0KZW5kb2JqCjkgMCBvYmoKPDwvRmlsdGVyIC9GbGF0ZURlY29kZSAvTGVuZ3RoIDE0Pj4Kc3RyZWFtCnic+///PwMDAwAO9wL+CmVuZHN0cmVhbQplbmRvYmoKMiAwIG9iago8PAovUHJvY1NldCBbL1BERiAvVGV4dCAvSW1hZ2VCIC9JbWFnZUMgL0ltYWdlSV0KL0ZvbnQgPDwKL0YxIDUgMCBSCi9GMiA2IDAgUgovRjMgNyAwIFIKPj4KL1hPYmplY3QgPDwKL0kxIDggMCBSCj4+Cj4+CmVuZG9iagoxMCAwIG9iago8PAovUHJvZHVjZXIgKEZQREYgMS42KQovQ3JlYXRpb25EYXRlIChEOjI [... cut ...] FIKL0luZm8gMTAgMCBSCj4+CnN0YXJ0eHJlZgo1Njk2CiUlRU9GCg==";
    console.log("Initialize convert Base 64...");
    let bytes =  this.print.base64ToArrayBuffer(base64);
    console.log(bytes);
    this.bytesConverted = bytes;
  }
```

config.page.html:

```
<ion-button expand="full" color="primary" (click)="convertBase64()">
    Convert base64
  </ion-button>

  <ion-button expand="full" color="primary" (click)="printConverted()">
    Print Converted
  </ion-button>
```
This is the printer:

[image]

PauloMotaZetec, 2022-04-26:
Hi,
Can you print images? Does that printer allows it?
Probably you will have to convert the PDF to image, using ImageMagick.

nicoguevara, 2022-07-19:
Hello, I have the same problem, were you able to solve it?

Anne-TF, 2023-03-21:
Did someone find the solution? I'm struggling with this rn