don/BluetoothSerial #448: iOS - Device not found, but Device shows BT connected/paired?

allidoisace, 2022-04-04:
I am currently using Ionic 6 in React on iOS. I am really struggling to get this working as I receive a `Device not found`. The iOS device shows the interlocking chains at the top right with BT connected to the Device Name.

```
const selectAccessory = async () => {
    const deviceDiscovery = BluetoothClassicSerialPort.setDeviceDiscoveredListener().subscribe(async (connectionConfig) => {
      console.log(connectionConfig);
      deviceDiscovery.unsubscribe();
    })
    await BluetoothClassicSerialPort.discoverUnpaired().catch(error => {
      console.log(error);
      deviceDiscovery.unsubscribe();
    })
  }
```

The selection for the Accessory pops up with the device name on the app, I select it and receive the error `Device not found`. However, BT is connected to the device. Even if I run a `list()` after the fact, it is an empty array.

kieranbarlow, 2022-08-05:
Did you manage to get this working? I'm struggling to pick up any devices on IOS.

vitorcrivano, 2023-07-31:
For me its the same here.. It works like charm on Android but the same device doens't show on the list for iOS.
Any updates on this issue?