don/BluetoothSerial #450: I can't use the plugin in ionic + react

Lucas-Wanat, 2022-07-19:
There is no adapted or updated documentation, besides the repository updates being from a few years ago, I would like to be given updated and adapted documentation for capacitor and react as well. Thank you very much in advance

PS.: My English is not very good, I'm from Brazil and I'm using the translator.

paesedombre, 2022-08-04:
I'm using Ionic and Angular and I'm using this plugin for an app that connects with BLE devices.
Check this guide https://ionicframework.com/docs/native/bluetooth-serial, if you search for help contact me

Lucas-Wanat, 2022-08-25:
The problem was solved, I didn't know how to extract the data that came, but I still have another one, the connect function

closed 2022-08-25 by Lucas-Wanat