don/BluetoothSerial #451: BluetoothSerial.connect() doesn't work

Lucas-Wanat, 2022-08-25:
I'm doing an ionic project with react and capacitor, but the app just doesn't connect with other devices that I've already tested, like another android, a windows 10 laptop and an ESP32 (which is the device that the smartphone will connect to when using company). I've tried with BLE but I haven't had any success either.

WHdev711, 2022-10-19:
Hello
I have same issue.
I hope you fix it

Lucas-Wanat, 2022-11-29:
> Hello I have same issue. I hope you fix it

I ended up not being able to solve this problem, but the team decided to use ionic angular, with part of a code bought to use, and it works like in the documentations. So it must be the solution.