don/BluetoothSerial #454: Bluetooth Serial Permissions Issue on Android 12

bevinGithub, 2022-11-09:
I need assistance with the Bluetooth Serial plugin, when l check for bluetooth.isEnable it returns Ok or is Disabled which is correct.  If l try to use any other functions for example List or discoverUnpaired l get the error below
**ERROR**
"Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10467, packageName = com.fire.service.app, attributionTag = null, token = android.os.BinderProxy@a208f16, next = null }: Starting discovery."

Your assistance will be much appreciated.

Below is my environment
Ionic:
   Ionic CLI                     : 6.20.1 
   Ionic Framework               : @ionic/angular 6.2.2
   @angular-devkit/build-angular : 13.0.4
   @angular-devkit/schematics    : 13.0.4
   @angular/cli                  : 13.0.4
   @ionic/angular-toolkit        : 5.0.3

Cordova:
   Cordova CLI       : 11.0.0
   Cordova Platforms : android 10.1.2
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 14 other plugins)

Utility:
   cordova-res                          : 0.15.4
   native-run (update available: 1.7.1) : 1.5.0

System:
   NodeJS : v14.15.4 (C:\Program Files\nodejs\node.exe)
   npm    : 6.14.10
   OS     : Windows 10

lapisanlangit, 2022-11-13:
same with me.. any solution?

bevinGithub, 2022-11-14:
No response yet

lapisanlangit, 2022-11-17:
oh my god.. i need this solution soon

bevinGithub, 2022-11-17:
same here brother been almost a month researching aslcan connect phone to ble devices but l want phone to phone

jabiinfante, 2022-11-24:
Hi!

We have managed to have this plugin working on Android 12, etc...

The solution is yet not well tested but you might find it usefull.

Remember to add these permissions on your app's AndroidManifest.xml
```xml
  <!--BLUETOOTH PERMISSION-->
  <uses-permission android:name="android.permission.BLUETOOTH" tools:replace="android:maxSdkVersion" android:maxSdkVersion="30" />
  <!-- Needed only if your app looks for Bluetooth devices. If your app doesn't use Bluetooth scan results to derive physical location information, you can strongly assert that your app doesn't derive physical location. -->
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
       android:usesPermissionFlags="neverForLocation"
       tools:targetApi="s" />
  <!-- Needed only if your app communicates with already-paired Bluetooth devices. -->
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
  <!-- Needed only if your app uses Bluetooth scan results to derive physical location. -->
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

```

[cordova-plugin-bluetooth-serial+0.4.7.patch.zip](https://github.com/don/BluetoothSerial/files/10086498/cordova-plugin-bluetooth-serial%2B0.4.7.patch.zip)

(The patch is generated and can be applied using [patch-package](https://www.npmjs.com/package/patch-package).

brandomcombr, 2022-11-25:
I change the AndroidManifest.xml with this suggestions, but tags tools:replace="android:maxSdkVersion" and tools:targetApi="s", doesnt find in my project, after removing this tags my project starting in debug mode, but when you start bluetooth the error appears
`I/ViewRootImpl@1a9ba7c[MainActivity]: stopped(false) old=false
D/CordovaActivity: Resumed the activity.
E/PluginManager: Uncaught exception from plugin
    java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{1a7314 513:br.com.brandom.comandacloud/u0a632} (pid=513, uid=10632) requires android.permission.BLUETOOTH_CONNECT
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2438)
        at android.os.Parcel.createException(Parcel.java:2422)
        at android.os.Parcel.readException(Parcel.java:2405)
        at android.os.Parcel.readException(Parcel.java:2347)
        at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2897)
        at android.app.Instrumentation.execStartActivity(Instrumentation.java:1743)
        at android.app.Activity.startActivityForResult(Activity.java:5473)
        at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:597)
        at org.apache.cordova.CordovaActivity.startActivityForResult(CordovaActivity.java:359)
        at android.app.Activity.startActivityForResult(Activity.java:5431)
        at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:583)
        at org.apache.cordova.CordovaInterfaceImpl.startActivityForResult(CordovaInterfaceImpl.java:69)
        at com.megster.cordova.BluetoothSerial.execute(BluetoothSerial.java:214)
        at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
        at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
        at org.apache.cordova.PluginManager.exec(PluginManager.java:146)
        at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
        at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:335)
        at android.os.Looper.loopOnce(Looper.java:186)
        at android.os.Looper.loop(Looper.java:313)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: android.os.RemoteException: Remote stack trace:
        at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1334)
        at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:1275)
        at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:906)
        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1886)
        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1757)
I/chromium: [INFO:CONSOLE(43427)] "ERROR Error: Uncaught (in promise): Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{1a7314 513:br.com.brandom.comandacloud/u0a632} (pid=513, uid=10632) requires android.permission.BLUETOOTH_CONNECT", source: https://localhost/vendor-es2015.js (43427)
I/IMM_LC: hsifw() - flags=0, caller=android.view.inputmethod.InputMethodManager.hideSoftInputFromWindow:1858 org.chromium.content.browser.input.ImeAdapterImpl.f:40 org.chromium.content.browser.input.ImeAdapterImpl.updateState:32 android.os.MessageQueue.nativePollOnce:-2 android.os.MessageQueue.next:335 
I/IMM_LC: hideSoftInputFromWindow - mService.hideSoftInput`

brandomcombr, 2022-11-25:
I create a fork this repo and do changes necessary for working in my project
https://github.com/brandomcombr/BluetoothSerial

lapisanlangit, 2022-11-27:
> I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial

awesome this is working!!!.. i have searched all cordova bluetooth library and tried it..but no luck.. thank you very much @brandomcombr

giuseppelanzi, 2022-12-04:
Coincidentally i was doing a very similar thing of @brandomcombr, in my fork https://github.com/giuseppelanzi/BluetoothSerial.

I'm posting this message only because I added something. 
I don't like the fact that the plugin is always asking for fine location permission. That should be used if I need to determine the physical position of the device, which I don't. So I added the boolean parameter **fineLocation** to **discoverUnpairedDevices**.

Hope it helps if someone is struggling for a similar reason.

lapisanlangit, 2022-12-11:
> > I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial
> 
> awesome this is working!!!.. i have searched all cordova bluetooth library and tried it..but no luck.. thank you very much @brandomcombr

i have found new problem again.. if use device with **Android 12 => working properly** but if used device **Android 11** getting error "error search location permission is required to discover unpaired device"

I also tried  https://github.com/giuseppelanzi/BluetoothSerial .. but still have same error.. problem is using Android 11

shoerofi, 2022-12-16:
> > > > I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial
> > > 
> > > 
> > > awesome this is working!!!.. i have searched all cordova bluetooth library and tried it..but no luck.. thank you very much @brandomcombr
> > 
> > 
> > i have found new problem again.. if use device with Android 12 => working properly but if used device Android 11 getting error "error search location permission is required to discover unpaired device"
> 
> @lapisanlangit, please try [brandomcombr#1](https://github.com/brandomcombr/BluetoothSerial/pull/1)

yeah, this is working on my android 11. thanks

lajith111, 2022-12-17:
> I need assistance with the Bluetooth Serial plugin, when l check for bluetooth.isEnable it returns Ok or is Disabled which is correct. If l try to use any other functions for example List or [...]

> I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial

How can I use this in my capacitor project...please let me know

emilkitua, 2023-03-05:
> yeah, this is working on my android 11. thanks

How did you implement this on your project? I am still facing the challenge. I even replaced the package with (https://github.com/brandomcombr/BluetoothSerial/pull/1) still doesn't work

shoerofi, 2023-03-05:
> How did you implement this on your project? I am still facing the challenge. I even replaced the package with (https://github.com/brandomcombr/BluetoothSerial/pull/1) still doesn't work

I am using cordova. I just change the git url in package.json
In my capacitor project I use the original plugin. No problem at all.

emilkitua, 2023-04-26:
i made the change as suggested and it worked. but this week when updating my app to support android 13 it stopped working.

it is giving me an error:
Cannot find module '@awesome-cordova-plugins/bluetooth-serial' or its corresponding type declarations.

my dependencies in package.json

"dependencies": {
    "@angular/common": "^14.0.0",
    "@angular/core": "^14.0.0",
    "@angular/forms": "^14.0.0",
    "@angular/platform-browser": "^14.0.0",
    "@angular/platform-browser-dynamic": "^14.0.0",
    "@angular/router": "^14.0.0",
    "@auth0/angular-jwt": "^5.1.0",
    "@awesome-cordova-plugins/bluetooth-serial": "github:brandomcombr/BluetoothSerial",
    "@awesome-cordova-plugins/core": "^6.0.1",
    "@awesome-cordova-plugins/device": "^6.0.1",
    "@awesome-cordova-plugins/file": "^6.3.0",

my import in app.module.ts

import { SQLitePorter } from '@awesome-cordova-plugins/sqlite-porter/ngx';
import { BluetoothSerial } from '@awesome-cordova-plugins/bluetooth-serial';
import { PDFGenerator } from '@ionic-native/pdf-generator/ngx';

Please help @shoerofi

emilkitua, 2023-04-30:
> I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial

how can I replace the current plugin with yours in my project?

Deliana90, 2023-09-16:
anyone have a full example of how to use this with angular? ^_^ thanks

Deliana90, 2023-11-06:
> > > > > > > I create a fork this repo and do changes necessary for working in my project https://github.com/brandomcombr/BluetoothSerial
> > 
> > 
> > > > > >
> > 
> > 
> > > > > >
> > 
> > 
> > > > > > awesome this is working!!!.. i have searched all cordova bluetooth library and tried it..but no luck.. thank you very much @brandomcombr
> > 
> > 
> > > > >
> > 
> > 
> > > > >
> > 
> > 
> > > > > i have found new problem again.. if use device with Android 12 => working properly but if used device Android 11 getting error "error search location permission is required to discover unpaired device"
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > @lapisanlangit, please try [brandomcombr#1](https://github.com/brandomcombr/BluetoothSerial/pull/1)
> > 
> > 
> > > 
> > 
> > 
> > > yeah, this is working on my android 11. thanks
> > 
> > 
> > How did you implement this on your project? I am still facing the challenge. I even replaced the package with ([brandomcombr#1](https://github.com/brandomcombr/BluetoothSerial/pull/1)) still doesn't work
> 
> I am using cordova. I just change the git url in package.json In my capacitor project I use the original plugin. No problem at all.

what package.json looks like, and how to import and use it. example code pls.

lajith111, 2024-04-10:
Some one please share package.json for  link to https://github.com/brandomcombr/BluetoothSerial/pull/1

so that i canuse in adroid 13

emilkitua, 2024-04-10:
during installation

use:  `ionic cordova plugin add https://github.com/brandomcombr/BluetoothSerial`
instead of `ionic cordova plugin add cordova-plugin-bluetooth-serial`

lajith111, 2024-04-10:
Thank you very much  @emilkitua .. since long time I am manually coping

lajith111, 2024-04-12:
@emilkitua  i am using capacitor? is it works on capacitor?