don/BluetoothSerial #455: App crash after start, if I use this plugin

SteveeH, 2022-11-19:
Hello, 

i want to use this plugin in my application, I just create new Cordova but after I install it "cordova plugin add cordova-plugin-bluetooth-serial" application always crash after launch, even if I didn't use any code from plugin (its just helloWorld app). I want to add some logs, but I can't open debug console because app crash immediately.

Do you have any idea, is it because of new version of iOS?

Cordova: 11.0.0
iOS : 16.1

Thanks you all for response

Glavotaner, 2023-03-06:
A good start is to check what the XCode console says before the app crashes; there you should be able to find the error that caused it. There is definitely initialization logic that runs on the plugin regardless if you actually invoked any method.

pitamash, 2023-06-22:
calling this bluetoothSerial.isConnected caused it to crash for me.

Glavotaner, 2023-06-22:
> calling this bluetoothSerial.isConnected caused it to crash for me.

@pitamash What platform are you on? Can you try to copy-paste the error you get from logcat or Xcode console?

pitamash, 2023-06-22:
I added this in androidManifest.xml file;

<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
android:usesPermissionFlags="neverForLocation" />

Glavotaner, 2023-06-22:
@pitamash You won't be able to use the new SCAN/CONNECT/DISCOVER permissions; you can put them in your manifest, but they are runtime permissions so the plugin still has to check/request those permissions at runtime. That won't happen because the project isn't maintained and while people have made PRs for that, the author isn't available to merge them.

I'd advise you to make your own fork, or use someone else's; eg. this one is mine https://github.com/Glavotaner/bluetooth-serial