don/BluetoothSerial #459: App crashed after call bluetoothSerial.connect()

LeDong95, 2023-03-23:
I tested with 2 phone, both have Android 13
Before (like a few weeks ago), the app i built run normal. 
But in past week, the app i build can't call bluetoothSerial.connect() function in my phone anymore.
It can call bluetoothSerial.list() function to get list of bonded device, but when i try to test, app crash instant, no error was show.
I tried to put console log in each step in my code (after subscribe to bluetoothSerial.connect() function) to debug , but nothing show in console

**Can anyone help me find where the issue is please?**

My phone: Android 13
My app ionic info:
Ionic:

   Ionic CLI                     : 6.20.8
   Ionic Framework               : @ionic/angular 6.5.2
   @angular-devkit/build-angular : 15.2.4
   @angular-devkit/schematics    : 15.2.4
   @angular/cli                  : 15.2.4
   @ionic/angular-toolkit        : 7.0.0

Cordova:

   Cordova CLI       : 11.1.0
   Cordova Platforms : android 10.1.2
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 4.2.1, (and 14 other plugins)

Utility:

   cordova-res : 0.15.4
   native-run  : 1.7.1

System:

   Android SDK Tools : 26.1.1
   NodeJS            : v18.12.1
   npm               : 9.4.1
   OS                : Windows 10

LeDong95, 2023-03-23:
> You need to target API 30. If you target 31 or higher this plugin will not work.

My friend phone using android 11, he can use app normal
If i target api 30, i can't publish app to google store, it require api >= 31, that why i target api 33

giuseppelanzi, 2023-03-30:
I forked the repot because the master fork has a lot of problem. You can try to use https://github.com/giuseppelanzi/BluetoothSerial or one the other forks that you'll find in the recent issues (especially https://github.com/don/BluetoothSerial/issues/454). 

Hope it helps.

LeDong95, 2023-03-30:
Thanks for your help.
I don't know what happen, but here is how i fixed it: I change some dependence version in package.json file
Old package.json
"dependencies": {
    "@angular/common": "^15.2.3",
    "@angular/core": "^15.2.3",
    "@angular/forms": "^15.2.3",
    "@angular/localize": "^15.2.3",
    "@angular/platform-browser": "^15.2.3",
    "@angular/platform-browser-dynamic": "^15.2.3",
    "@angular/router": "^15.2.3",
    "@capacitor/core": "4.5.0",
    "@capacitor/keyboard": "4.1.0",
    "@capacitor/status-bar": "4.1.0",
    "@ionic/angular": "^6.5.2",
    "@ionic/cordova-builders": "^7.0.0",
    "rxjs": "^6.6.0",
    ...
},
"devDependencies": {
    "@angular-devkit/build-angular": "^15.2.4",
    "@angular-eslint/builder": "^14.0.0",
    "@angular-eslint/eslint-plugin": "^14.0.0",
    "@angular-eslint/eslint-plugin-template": "^14.0.0",
    "@angular-eslint/template-parser": "^14.0.0",
    "@angular/cli": "^15.2.4",
    "@angular/compiler": "^15.2.3",
    "@angular/compiler-cli": "^15.2.3",
    "@angular/language-service": "^15.2.3",
    "cordova-plugin-ionic-webview": "^4.0.0",
    "karma": "^6.4.1",
    "karma-jasmine": "~3.3.0",
    "karma-jasmine-html-reporter": "^1.5.0",
    ...
}

New package.json
"dependencies": {
    "@angular/common": "^15.0.0",
    "@angular/core": "^15.0.0",
    "@angular/forms": "^15.0.0",
    "@angular/localize": "^15.0.0",
    "@angular/platform-browser": "^15.0.0",
    "@angular/platform-browser-dynamic": "^15.0.0",
    "@angular/router": "^15.0.0",
    "@capacitor/core": "4.7.1",
    "@capacitor/keyboard": "4.1.1",
    "@capacitor/status-bar": "4.1.1",
    "@ionic/angular": "~6.6.3",
    "@ionic/cordova-builders": "^8.0.0",
    "rxjs": "^6.5.3",
    ...
},
"devDependencies": {
    "@angular-devkit/build-angular": "^15.0.0",
    "@angular-eslint/builder": "^15.0.0",
    "@angular-eslint/eslint-plugin": "^15.0.0",
    "@angular-eslint/eslint-plugin-template": "^15.0.0",
    "@angular-eslint/schematics": "^15.0.0",
    "@angular-eslint/template-parser": "^15.0.0",
    "@angular/cli": "^15.0.0",
    "@angular/compiler": "^15.0.0",
    "@angular/compiler-cli": "^15.0.0",
    "@angular/language-service": "^15.0.0",
    "cordova-plugin-ionic-webview": "^5.0.0",
    "karma": "~6.4.0",
    "karma-jasmine": "~5.1.0",
    "karma-jasmine-html-reporter": "~2.0.0",
    ...
}

I don't know why it fixed, but after change package.json file like this and build app. My app working normal again
(The reason i change like that is i create a new app with ionic command, add Bluetooth Serial plugin, change some sample code i found online, build app. This new app worked in my phone, so i change some dependence in my app to similar to this new app)